ruuvi/ruuvi.drivers.c #184: LEDs are not tested - defect

DG12, 2021-02-09:
No test manager exists for LEDs :
ruuvi.drivers.c/test/tasks/test_ruuvi_task_led.c includes:
setUp, tearDown, test_rt_led_init_leds_success_gpio_was_init, test_rt_led_init_leds_gpio_not_init,
test_rt_led_init_leds_twice_fails, test_rt_led_init_leds_gpio_fails, test_rt_led_write_not_init, test_rt_led_write_not_led
test_rt_led_write_valid, test_rt_led_activity_led_set_valid, test_rt_led_activity_led_set_unused ,
test_rt_led_activity_led_set_invalid, test_rt_led_is_led, test_rt_led_blink_start_ok , test_rt_led_blink_start_not_init
test_rt_led_blink_start_timer_fail , test_rt_led_blink_start_led_blinking, test_rt_led_blink_start_not_led ,
test_rt_led_blink_stop_ok  and test_rt_led_blink_isr 
but does not include a function to call them.

Maybe the test_LED function should be included in 
ruuvi.drivers.c/src/integration_tests/ruuvi_interface_gpio_test.c

ojousima, 2021-02-09:
Led task coverage report is here: https://sonarcloud.io/component_measures?id=ruuvi_ruuvi.drivers.c&metric=coverage&selected=ruuvi_ruuvi.drivers.c%3Asrc%2Ftasks%2Fruuvi_task_led.c&view=list
Coverage is 97.8% which exceeds the target of 80%.

closed 2021-02-09 by ojousima

DG12, 2021-02-09:
There is no observable difference between test and default with respect to LEDs.

ojousima, 2021-02-10:
That would be a matter for integration test of firmware, not drivers. And LEDs are simple devices, if GPIO test passes it can be safely assumed that LEDs connected to GPIOs will work as far as firmware is concerned. 

Factory testing assembled boards is a separate matter and outside the scope of drivers or Ruuvi Firmware.

DG12, 2021-02-10:
Perhaps ruuvi.drivers.c/test/tasks/test_ruuvi_task_led.c  should be deleted?

ojousima, 2021-02-10:
> Perhaps ruuvi.drivers.c/test/tasks/test_ruuvi_task_led.c should be deleted?

No. It tests the code related to leds.