ruuvi/ruuvi.drivers.c #198: Unused file should be deleted: ruuvi_driver_test.c

DG12, 2021-03-07:
Please delete unused  file. 
ruuvi.drivers.c/src/integration_tests/ruuvi_driver_test.c
The function  rl_expect_close is included in ruuvi_library_test.c (notice function. named here  letter L

```
bool ri_expect_close (const float expect, const int8_t precision,
                      const float check)
{
    if (!isfinite (expect) || !isfinite (check)) { return false; }
    const float max_delta = pow (10, precision);
    float delta = expect - check;
    if (delta < 0) { delta = 0 - delta; } // absolute value
    return max_delta > delta;
}

```

ojousima, 2021-03-08:
Since ruuvi libraries are not a part of Ruuvi drivers, the file is not unused. I suppose we could pass the comparison function as a function pointer, but that's the kind of micro-optimization of memory footprint I really would like to avoid unless absolutely necessary.

closed 2021-03-08 by ojousima

DG12, 2021-03-08:
Can you please explain this a little further.
Although the load map shows ruuvi_driver_test.c.o the text, data and bss sections are 0 length.
The symbol ri_expect_close does not occur in the load map and if I add a call to it  loading complains with
main.c:122: undefined  reference to `ri_expect_close'
collect2: error: ld returned 1 exit status

PS I prefer to leave optimization to the experts (i.e. compiler and loader)